DF4IAH/RedPitaya_RadioBox #80: Front-end: entering frequency needs a usable controller

DF4IAH, 2016-01-22:
Entering an exact frequency is OK for diagnostic, but for better user interaction another __controller box__ is needed for fast and easy frequency change.

___These items shall the frequency editing controller box include___:
- [ ] any frequency field should be editable by that frequency controller box when clicked into. Behaviour like a radio button, so one frequency edit box receives all user interaction for any frequency edit field
- [ ] each digit an up button above the digit, a down button below the digit
- [ ] over- and underflow of each digit shall change the next digit left of it
- [ ] when available a mouse wheel shall be connected to the up and down buttons to each digit where the mouse is positioned over
- [ ] each digit, and only that digit, shall be highlighted when clicked to, after that a slider activates up/down clicks/second for easy frequency scanning

ted051, 2016-02-29:
Please have a look at:
[http://www.qrpforum.de/index.php?page=Attachment&attachmentID=16124&h=789c097234bb0c927065fca74a28ee26a0061123]
Independent from you I had the same idea...
The hardware is a STM32F7 Disco board with touchscreen 480 x 272. Connected to RedPitaya via UART 115200 Baud.

DF4IAH, 2016-03-14:
Thank you, I had a similar idea to do it. You will find the button-set nearly same arranged as your linked picture.

Until now I have a HTML/CSS body. Next I have to put life into HTML/JavaScript for the user interactions.

DF4IAH, 2016-03-20:
Done with c28b3f62563c913f2c06e214b0689ec0c6ae7e46

closed 2016-03-20 by DF4IAH