DF4IAH/RedPitaya_RadioBox #91: TX: add output gain correction for 15 MHz flaw

DF4IAH, 2016-02-29:
The RedPitaya DAC are followed by low pass filters. At abt. 15 MHz there seems to be a pole which extends the signal to a magnification value of abt 10.

More to be analyzed and enhanced, here.

pavel-demin, 2016-02-29:
Out of curiosity. Is it really important to correct this?

I've also noticed that Red Pitaya output without a 50 Ohm load resonates at about 15 and 45 MHz.

However, with a 50 Ohm load, the frequency response is quite flat as you can see, for example,  on the first plot in the following comment:
https://github.com/pavel-demin/red-pitaya-notes/issues/170#issuecomment-174176177

DF4IAH, 2016-02-29:
Thank you for your idea Pavel. Indeed with a 50 ohms resistor the gain is nearly flat. That seems not to be the problem. I know the HAM radio operators somehow and I am sure, most of them do not connect a 50 ohms resistor when connecting to anywhere.

My idea is to have a table of correction values for both, with and without a 50 ohms load. In my application front-end I will have a switch to activate the "50 ohms" output, defaulting to "open end". So there would be never more voltage at the output as intended. No OM that would make me responsible to shoot anything. Who knows to set the switch drives the high power with the "50 ohms" connected.

The poles at 17 MHz gives me a 14x overshoot voltage when the end is not terminated - the other at 46 MHz still gives me a factor of 5x.

Of course, that is not totally fool proov but it gives me more chance to hold of those guys.  :-)

Greetings
  Uli

pavel-demin, 2016-02-29:
Thank you for the explanation. I did not realize that it could damage some external equipment. Now, I should think of how to add an output gain correction to my SDR applications.

DF4IAH, 2016-03-11:
Fixed with 9f0b3dce180eb5b5ef70f87a5bd1c441dbe8c7c8

closed 2016-03-11 by DF4IAH