DF4IAH/RedPitaya_RadioBox #143: RadioBox_Upgrade.sh fail on 095-1

effeerre75, 2016-06-03:
Full log is attached

Step 1:
------
mv: cannot stat ‘/lib/modules’: No such file or directory
rm: cannot remove ‘/lib/modules_old’: No such file or directory
depmod: ERROR: could not open directory /lib/modules/3.18.0-xilinx: No such file or directory
depmod: FATAL: could not search modules: No such file or directory
... done.
------

Step 2: no problem

Step 3: no problem

Step 4:
      - jackd2 configuration: answered "no" at realtime priority 
      - icecast2 configuration: answered "yes" and leaved all as default 

Step 5: no problem

Step 6: Failed to fetch fldigi from http://ppa.launchpad.net

Step 7: no problem

Step 8:
------
/opt/redpitaya/www/apps/radiobox/bin/RadioBox_Upgrade.sh: 58: /opt/redpitaya/www/apps/radiobox/bin/RadioBox_Upgrade.sh: cannot create data/RadioBox-Upgrade_dpkg-selections-current.dat: Directory nonexistent
cat: data/RadioBox-Upgrade_dpkg-selections-needed.dat: No such file or directory
cat: data/RadioBox-Upgrade_dpkg-selections-current.dat: No such file or directory
/opt/redpitaya/www/apps/radiobox/bin/RadioBox_Upgrade.sh: 59: /opt/redpitaya/www/apps/radiobox/bin/RadioBox_Upgrade.sh: cannot create data/RadioBox-Upgrade_dpkg-selections-new.dat: Directory nonexistent
/opt/redpitaya/www/apps/radiobox/bin/RadioBox_Upgrade.sh: 60: /opt/redpitaya/www/apps/radiobox/bin/RadioBox_Upgrade.sh: cannot open data/RadioBox-Upgrade_dpkg-selections-new.dat: No such file
... done.
------

Step 9: no problem

Step 10: no problem

Step 11: no problem

Step 12:
------
cp: cannot stat ‘/opt/redpitaya/etc/xinetd.conf’: No such file or directory
cp: cannot stat ‘/opt/redpitaya/etc/xinetd.d’: No such file or directory
... done.
------

Step 13: 
------
mv: cannot stat ‘/etc/ices2’: No such file or directory
cp: cannot stat ‘/opt/redpitaya/etc/pulse’: No such file or directory
cp: cannot stat ‘/opt/redpitaya/etc/icecast2’: No such file or directory
cp: cannot stat ‘/opt/redpitaya/etc/ices2’: No such file or directory
chown: cannot access ‘/etc/icecast2’: No such file or directory
chown: cannot access ‘/etc/ices2’: No such file or directory
------

Step 14:
/opt/redpitaya/www/apps/radiobox/bin/RadioBox_Upgrade.sh: 115: /opt/redpitaya/www/apps/radiobox/bin/RadioBox_Upgrade.sh: redpitaya-ac97_stop: not found
cp: cannot stat ‘data/RadioBox-Upgrade_asound.state’: No such file or directory
/opt/redpitaya/www/apps/radiobox/bin/RadioBox_Upgrade.sh: 117: /opt/redpitaya/www/apps/radiobox/bin/RadioBox_Upgrade.sh: redpitaya-ac97_start: not found
alsactl: load_state:1677: Cannot open /var/lib/alsa/asound.state for reading: No such file or directory
Connection failure: Connection refused
pa_context_connect() failed: Connection refused
Connection failure: Connection refused
pa_context_connect() failed: Connection refused
Connection failure: Connection refused
pa_context_connect() failed: Connection refused
... done.

Step 15: no problem

[RadioBox_Upgrade_log.txt](https://github.com/DF4IAH/RedPitaya_RadioBox/files/297335/RadioBox_Upgrade_log.txt)

DF4IAH, 2016-06-06:
Thank you very much for your log. It seems that I still have to do much on it.

For Step 1 it works when rebooted after the Update script again. That is due to the fact that the current RedPitaya builds a 4.0 kernel but the filesystem is still based on the 3.18.0 kernel. After a reboot this message should go away and the current module directory being set to the right place.

Step 4: it is all "OK" for answering, prefered are "no" and "no" for these two pop-ups. The kernel seems to run safe when the realtime option is disabled.

Step 6: also I have noticed it, but it results to a working fldigi program that works anyway.

All other Steps should work when started again after a reboot. Can you please give it a try again?

Thank you very much for your assistance.
Uli, DF4IAH

effeerre75, 2016-06-07:
Hi, just made a clean install of the RedPitaya system just for be sure.
Lounched update, rebooted and updated again. Some errors gone and some other come up.
Unfortunally still not working.
Thanks for this great work!
[RadioBox_Upgrade_log.txt](https://github.com/DF4IAH/RedPitaya_RadioBox/files/302590/RadioBox_Upgrade_log.txt)

DF4IAH, 2016-06-07:
Thank you for your time and effort you have taken and sorry for that.

RadioBox does work without the upgrade already. So you can listen to the radio and go on air without that upgrade. The upgrade does install the icecast2 server, so anybody connecting the device on port 8000 can listen to the same radioprogram - that's the upgrade. And the processor running at 666 MHz instead of 500 MHz, but that is not really needed.

Just for being sure: your RadioBox screen within the webbrowser comes up and you can connect speakers at the two SMA output connectors to start listening? If not, we should look first there.

Please tell me if you intend to upgrade to the streaming server or you are just at the beginning having  problems to receive anything.

Meanwhile I do make reworks on the Upgrade script to make it more robust.

Thank you alot and sorry for any inconvinience,
  Uli

effeerre75, 2016-06-07:
Hi, don't worry, things that happen.
Radiobox "standard" RX works really fine! Just made an adapter cable for the 4pin iPhone earphone that have mic. Tomorrow I'll try also TX.

Audio streaming server would be great for connecting to, for example, an iPad and doing all from remote via browser.

73 IW2MVI

DF4IAH, 2016-06-07:
Great!  I think I have it solved much better now with 605526292835f162acb3a4e247476ab1a7addb42 .
The new version of the App is also available at the Bazaar, now.

Thank you for your assistance!

Vy 73
  Uli, DF4IAH

effeerre75, 2016-06-08:
That worked great! :)
Just a system hang after it finished the second update but a power off/on solved it.

Now I can start streaming from Firefox no way from Safari, audio format not supported? It's OGG Vorbis?
The audio level seems a little low: in the first screenshot the output of RadioBox and in the second the output of GQRX. Stations is DDK9 on 10.100Mhz. There is always, also, a strong signal in the center, no matter the frequency or the mode.

RadioBox
[image]

GQRX
[image]

By the way you are made a really great work!

DF4IAH, 2016-06-09:
Yes, it is Ogg/Vorbis streaming, as RAW encoding is less supported by other browsers. Maybe I find another type instead for "application/ogg". Any suggestions are welcome. You can even open the icecast2 webserver on port 8000.

The low signal is due to the missing AGC units, what is planned to be added next. With the help of it an additional offset correction would be included, also. Therefore, the missing DC component would result in an reduced waver frequency leakage (1700 Hz).

I try to get also some RX stations at the long antenna and will have a look on your findings.

Thank you very much for your report.

73
  Uli, DF4IAH

DF4IAH, 2016-06-11:
In the meanwhile I fixed two more bugs in the ADC and DAC offset corrections what reduces some of the noise.

To take away the 1700 Hz noise I would have to replace the SSB weaver method by the Hilbert transformation method due to the fact, that the red pitaya hardware does have a poor power supply. Any noise by changing currents within the FPGA results in an DAC bounce of some millivolts. That explains also why a smart card access makes noise at the DAC SMA connectors.

Next week I find some time to build a new version for the Bazaar, that would fix some of this issues.

Cheers
  Uli

effeerre75, 2016-06-11:
Seems you are woking hard on this: really happy!
According to this paper the best combination for covering all the browser is with OGG Vorbis + AAC.
http://www.ibm.com/developerworks/library/wa-ioshtml5/index.html

Tried also to connect via VLC and Fstream on OSX: both works great using as source http://192.168.123.73:8000/redpitaya.raw
Played also a bit with the equalizer of fstream for amplify the signal: it's a little better.

Have a nice weekend!

DF4IAH, 2016-06-13:
Thank you! I'll have an eye on it for inclusion. This week I am busy but next weekend is in sight  :-)

DF4IAH, 2016-06-13:
Thank you! I'll have an eye on it for inclusion. This week I am busy but next weekend is in sight :-)
I would like to continue that conversiation in #149 and closing this packaging thing.

closed 2016-06-13 by DF4IAH